ysl208/iRoPro #59: Unable to find either executable 'empy' or Python module 'em'...  try   installing the package 'python-empy'

ysl208, 2021-04-17:
When running `catkin build` I get the following error
```
CMake Error at /opt/ros/melodic/share/catkin/cmake/empy.cmake:29 (message):
  Unable to find either executable 'empy' or Python module 'em'...  try
  installing the package 'python-empy'
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/all.cmake:163 (include)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:56 (find_package)
```

ysl208, 2021-04-18:
This can be solved by telling it to use python3
```
catkin build -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.7m
```

closed 2021-04-18 by ysl208

AnshShah3009, 2022-11-12:
Why doesn't it default to python3?

AbdelrahmanAbounida, 2022-12-17:
catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3

Tim-Liu-1, 2023-08-13:
> `catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3`

Thanks! Your solution works for me!

AwesomePeiweiPan, 2023-08-26:
> `catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3`

Yes, work for me too. Thanks a lot!

Jeremiah-HL, 2024-02-04:
That works for me, too! Thanks!